moodlebox/moodlebox #26: caractères incorrects dans la page d'authentification

christian-westphal, 2017-08-23:
Lors de la première connexion, la page d'authentification comporte des caractères incohérents dans le champ "nom d'utilisateur"
En utilisant la case "se souvenir etc" le problème ne se pose plus par la suite.

[image]

(à classer dans les pas grave mais pas bô)

martignoni, 2017-08-23:
Je n'arrive pas à reproduire le bogue. Tu peux indiquer l'environnement complet de ton client (navigateur, OS, etc.) ?

Est-ce que ça arrive sur d'autres configurations clientes ?

christian-westphal, 2017-08-23:
vu sur android 4.4 + firefox 55.0.2 et debian jessie + firefox 45.9.0
Je regarde avec ce que j'ai sous la main, le problème est que cela ne se passe qu'a la première connexion

christian-westphal, 2017-08-23:
Cela ne le fait pas avec un seamonkey *que je n'ai jamais utilisé avec la moodlebox* et ça ne le fait pas avec FF en navigation privée. En purgeant le cache et les cookies, cela ne le fait pas non plus. Du coup je pense à un pb de cache qui provient de la mémorisation de l'ancien compte connecté sur l'ancienne moodlebox. 
Désolé de t'avoir dérangé pour ça... j'essayerai de faire mieux à l'avenir avec un vrai bug  ;-)

martignoni, 2017-08-24:
Je pensais que c'était qqch comme ça.

Pas de dérangement : mieux vaut une annonce de trop que de pas assez. Je ferme le bogue.

closed 2017-08-24 by martignoni