moodlebox/moodlebox #30: Élargir la plage d'IP servie par le DHCP

christian-westphal, 2017-09-13:
Actuellement la plage d'IP servie par le DHCP est assez restreinte or dans le cas d'une utilisation en milieu scolaire avec un grand nombre de classe, elle peut-être insuffisante.
Exemple perso : 5 classes de 30 élèves, même si je n'ai pas toutes les classes la même journée, la moodlebox sert des IP aux portables allumés dans les salles voisines et les derniers portables ne se connectent plus.
J'ai modifié le réglage du DNSmasq.conf pour couvrir la plage 10.0.0.10 à 10.0.0.250 et ça semble OK.

Offrir la possibilité de régler ça par GUI ne me semble pas nécessaire mais modifier le réglage par défaut pourrait être utile (et/ou raccourcir la durée du bail ?)

Merci
Christian

martignoni, 2017-09-13:
Good idea. Will be there in next version.

martignoni, 2017-09-13:
I'll reduce the lease time to 6 hours too, so that unused IP addresses can be reused.

commit 84796ea1 (2017-09-13): Extend DHCP range and reduce lease time to enable more IP addresses.
Fixes #30.
closed 2017-09-13 by martignoni