moodlebox/moodlebox #32: Provide an administrative access to the MoodleBox from the Internet

martignoni, 2017-09-19:
This idea was [suggested by Michel Denis](https://moodle.org/mod/forum/discuss.php?d=358700#p1446690).

Could be done via VPN or other network tricks, e.g. SSH tunneling.

martignoni, 2017-09-19:
Alternative to investigate: [PageKite](https://pagekite.net).

michel-denis, 2017-09-21:
Intéressant ! A ton avis cela sera-t-il assez performant pour que le GUI de Moodle tourne bien ainsi ?

Merci encore,
-michel

ralf-krause, 2017-09-21:
Hi Nicolas,
it's no problem to access the MoodleBox via the ethernet cable. If you know the ip address you can login with ssh.

```
iMac-27:~ krause$ ssh moodlebox@192.168.0.10
The authenticity of host '192.168.0.10 (192.168.0.10)' can't be established.
ECDSA key fingerprint is SHA256:moYiKZPPwnkY5Azq8R6ogrxLO52koVbl4wFmtVQshh0.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.0.10' (ECDSA) to the list of known hosts.
moodlebox@192.168.0.10's password: 
Linux moodlebox 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Sep 13 23:23:52 2017 from 10.0.0.153
moodlebox@moodlebox:~ $ ls -al
insgesamt 32
drwxr-xr-x 4 moodlebox moodlebox 4096 Sep 12 22:15 .
drwxr-xr-x 3 root      root      4096 Sep 11 13:54 ..
-rw------- 1 moodlebox moodlebox  542 Sep 13 23:42 .bash_history
-rw-r--r-- 1 moodlebox moodlebox  220 Sep  7 16:59 .bash_logout
-rw-r--r-- 1 moodlebox moodlebox 3808 Sep 11 13:54 .bashrc
drwxr-sr-x 5 moodlebox www-data  4096 Sep 13 23:24 files
drwxr-xr-x 2 moodlebox moodlebox 4096 Sep 12 22:15 .nano
-rw-r--r-- 1 moodlebox moodlebox  675 Sep  7 16:59 .profile
moodlebox@moodlebox:~ $
``` 
But you open the MoodleBox with http even if you know the ip address. Moodle uses its config.php and tries to open http://moodlebox.home/ .... but there is no moodlebox.home in the dns server when you connect via ethernet lan. The router can't know that there is a MoodleBox with the ip address 192.168.0.10. Yes, if you set a static name for the ip address in your router then you can connect also from ethernet lan.

Ralf

martignoni, 2017-09-21:
@ralf-krause: sure, it's a built-in feature to access the MoodleBox this way from the same LAN.

The requested feature here is to access **from the Internet**, e.g. access from your place to **my** MoodleBox. It's another story :-)

martignoni, 2017-09-21:
> cela sera-t-il assez performant pour que le GUI de Moodle tourne bien ainsi ?

Impossible à dire: ça dépendra des réseaux disponibles.

ralf-krause, 2017-09-22:
@michel-denis,
if you only want to get the administration interface for the lan or for the internet it should not be very difficult.

If you want to get a full MoodleBox access from the lan then you have to define a static ip address with the dns name "moodlebox.home" in your router. You can't use the static ip address because the config.php has configured "moodlebox.home".

But it's not so easy to get the access from the internet because the dns name of the MoodleBox is a private name "moodlebox.home". If you want to use a dyndns service to get an offical name like "moodlebox.ddnss.ch" you need to register at the dyndns service and get the dns name. Then you must configure your router to use the dyndns service. Also you need to configure a port forwarding for port 80 to the static address of the MoodleBox. But this is not the only thing! You need to change the dns name inside the MoodleBox to "moodlebox.ddnss.ch". And you also need to change the www address in the Moodle config.php also to "moodlebox.ddnss.ch" and all internal links in Moodle. In this case you must access the MoodleBox with "moodlebox.ddnss.ch" also from the wlan.

I think that these changes should not be preconfigured in the MoodleBox. Nicolas should keep it simple.

Ralf

michel-denis, 2017-09-22:
Hi Ralf, thank you for your response and the info.

What we need is: from a node on the internet, to login as admin to that remote MoodleBox and perform typical Moodle Admin tasks using the Admin GUI.

-michel

ralf-krause, 2017-09-22:
@martignoni 
The reverse tunnel with SSH would be a nice idea forwarding the ports 22 and 80. Would it be possible to open the Moodle on the MoodleBox without any change in the config.php?

The settings in the router must be the same as I wrote before.

Ralf

martignoni, 2017-09-22:
> Would it be possible to open the Moodle on the MoodleBox without any change in the config.php?

It seems that Pagekite has an option to do this. I'm still testing for now.

BrunoMalaval, 2017-09-22:
Hello,

An alternative to Pagekite is ngrok (free).
I try it with my Moodlebox, it works
It can be launched with auth ( login/password )
It seems to be possible to start it as service, but don't try this yet.

Bruno

michel-denis, 2017-09-22:
hi Bruno, ngrok is free apparently in "exceptions" cases  - it's a little bit "hidden" at the bottom of the page : https://ngrok.com/pricing

Will it be enough in fonctionality and stability for operations mode ?
-michel

martignoni, 2017-09-22:
Thanks @BrunoMalaval for ngrok mention. I'll try. Yes @michel-denis it will be enough for such needs.

martignoni, 2017-09-22:
So I tested several alternatives. The most promising is ngrok. It works very well (with a free plan), but as Moodle URL is hard coded in its config file "config.php", it doesn't work fully correctly.

So I'm afraid it's not possible to do it cleanly without changing "config.php", and this would break the usage on the LAN.

I'll try to find any workaround, but don't hold your breath.

martignoni, 2017-09-22:
OK, I've found a workaround, but it's very hacky. It involves setting wwwroot in "config.php" by reading the default-server variable from nginx (at runtime), which defaults to the server hostname. It works, but I don't know what will happen with URLs in database.

I must say I'm very reluctant to add this as a default feature of the MoodleBox, as it could make more harm than help.

For reference, here's my hack of config.php:

```php
function wwwroot() {
    $wwwroot  = $_SERVER['HTTPS'] == 'on' ? 'https' : 'http';
    $wwwroot .= '://'.$_SERVER['HTTP_HOST'];

    return rtrim( $wwwroot, '/' );
}
$CFG->wwwroot   = wwwroot();
```

ralf-krause, 2017-09-22:
Hi Nicolas,
I think that Moodle can't work with different dynamic urls. Many urls for resources are saved in the database. If you will change the url of a moodle site you have to replace the urls with a Moodle tool. Call [http://moodlebox.home/admin/tool/replace/] to see what I mean.

martignoni, 2017-09-22:
Yes, that's why I wrote this: "It works, but I don't know what will happen with URLs in database." and why I'm so reluctant to do such a hack. Thanks for you input.

martignoni, 2017-09-23:
There seems to have no simple and non-hacky way to do this :-(. One has to

1. Have access to a public SSH server or have an account on Pagekit or Ngrok, **and**
2. Hack [the Moodle server in a very bad way](https://github.com/martignoni/make-moodlebox/issues/32#issuecomment-331529285) (thanks to @ralf-krause for confirmation of this).

I'll close this as "won't fix": anyway, if you're a nerd with enough knowledge to do the above steps, you can hack your own MoodleBox without difficulty.

Thanks to all commenters for their input.

closed 2017-09-23 by martignoni

frankkoch, 2018-07-30:
Hello Nicolas

Almost a year has passed since this discussion and I wonder if there is still no feasible way to do a remote login into a MoodleBox connected to the internet? I'm having a case in the Philippines with remote islands, where they have to ship the MicroSD-cards for maintenance...

Best, Frank

ralf-krause, 2018-07-30:
Hi Frank,
what do you want do do with the MoodleBox remote access? Do you want to connect to the ssh terminal? It is no problem to get the ssh terminal. But you can't connect to the MoodleBox via http.

You must set up a static ip address for the ethernet lan connection. Therefore you must configure the router dhcp service which sends the ip address on a restart of the MoodleBox. For this static ip address of the MoodleBox you must configure a port forwarding for the router firewall. And you need to configure a dynamic dns for the router if it isn't connected with a static ip address to the internet.

Let's use an example:
The MoodleBox has a static ip address 192.168.2.108 on the far island.
The router is connected to the internet and has the dynamic dns address moodlebox123.dyndns.org.
The port forward 2222 (extern) connects to the host 192.168.2.108 and port 22 (intern).
You can connect ssh directly to the remote MoodleBox.
Type the following: ssh moodlebox@moodlebox123.dyndns.org -p2222

The MoodleBox needs no extra configuration. 

Best regards, Ralf

frankkoch, 2018-07-30:
Hi Ralf

Thank you for the fast answer. We need remote access to maintain the MoodleBox. This might be due to  upgrade Moodle, or to install/change content within Moodle. I assume this can be done using ssh.

Another need might be to upgrade the MicroSD-card with a new MoodleBox version. Here I assume, this cannot be done using ssh.

Thank you also for the terminal instruction. I have to talk to a network admin for the static ip in order to try it.

Best, Frank