moodlebox/moodlebox #33: Change repository names to be more international

martignoni, 2017-09-20:
Proposed change:

- "Fichiers SFTP" -> "SFTP"
- "Clef USB" -> "USB"

ralf-krause, 2017-09-21:
What do you think about "USB Memory" and "SFTP Repository"?

martignoni, 2017-09-21:
I'd rather go with "USB" and "SFTP" (alone), as these are acronyms widely used in any language. But is it user friendly enough?

ralf-krause, 2017-09-21:
Yes, "USB" and "SFTP" are great! 

I thought that USB is a connection only so "USB Memory" tells something about the thing which should be connected ... USB stick, USB harddrive, USB ssd, USB SDCard with an adapter.

SFTP is a protocol. Perhaps you should call it "SFTP Files" because the folder is named "Files".

martignoni, 2017-09-22:
It seems that "USB Memory" and "SFTP Files" would do it, in French too. See https://moodle.org/mod/forum/discuss.php?d=358852#p1447497.

dajan, 2017-09-25:
Hello,
I would suggest, as I did in the Moodle forum, these two terms as a proposition 👍 
 
 USB Drive

because the term is understood and used in French, German, Italien but also in Spanish. This is also the term universally used by Google for its cloud storage service.

 Doc SFTP or Documents SFTP (with a preference for the former, because it is shorter and still meaningful)

Doc is the abbreviation of the English Document, the French Document, the Italian Documenti the Spanish Documento and it is close to the German Dokument.

IMHO

-dajan

martignoni, 2017-09-27:
Thank you for all your inputs (and forum discussions). I've decided to go with

- USB Drive
- SFTP Documents

Will be released in next image (no code change needed in this repo).

closed 2017-09-27 by martignoni