moodlebox/moodlebox #49: Moodle source installation directory is hardcoded

martignoni, 2017-12-03:
We have to move hardcoded `/var/www/moodle` installation directory to an ansible variable.

martignoni, 2017-12-03:
See commit 4c97212.

ralf-krause, 2017-12-05:
Hi Nicolas, 
as I know you will need a static url for Moodle. A lot of links for pictures and other resources are completely saved in the database. If you change the url of a Moodle platform you must replace all of these saved entries with a script. http://moodlebox.home/admin/tool/replace/
Best regards, Ralf

martignoni, 2017-12-05:
Thanks Ralf. This issue is about replacing the hardcoded values or the source and data directories in the building flow of the moodlebox (not replacing anything once the MoodleBox is installed. 

As you can see in commit 4c97212 , this is already done.

martignoni, 2017-12-08:
Tested, closing.

closed 2017-12-08 by martignoni