moodlebox/moodlebox #51: Adapt motd to display pertinent info

martignoni, 2017-12-12:
At login, mots should display info about MoodleBox version, author and copyright.

martignoni, 2017-12-22:
Very interesting resource about customizing `motd`: https://ownyourbits.com/2017/04/05/customize-your-motd-login-message-in-debian-and-ubuntu/.

martignoni, 2017-12-22:
See also: https://github.com/hopey-dishwasher/termpix

martignoni, 2017-12-26:
Fixed by 1d20bd0 and b784484. Closing.

closed 2017-12-26 by martignoni

ralf-krause, 2018-02-11:
I found that the logo doesn't look fine in my iPad Terminus app.
[image]

martignoni, 2018-02-11:
Indeed. Would you please send me the link to your terminal app?

ralf-krause, 2018-02-11:
Termius von Crystalnix
https://itunes.apple.com/de/app/termius/id549039908?mt=8

ralf-krause, 2018-04-01:
Hi Nicolas, I think that Terminus on iOS has problems with the inclused fonts. I tested Terminus on Android and the font PTmono shows the correct moodlebox logo. Ralf 
[image]

ralf-krause, 2018-07-11:
Hi Nicolas,
since today the Termius app for iOS shows the MoodleBox logo correctly. The Termius app is updated for iPhone and iPad. The app is free.
Ralf

[image]