moodlebox/moodlebox #57: Login Incorrect after first installation

lsteamgeo, 2018-02-01:
Bonjour,

Après transfert de l'image moodlebox-2.1.0.img.gz via Etcher sur une SDcard et lors du premier démarrage l'affichage de l'écran reste en ligne de commande. 

Un "login" et une "password" est demandé. 
_Raspbian GNU/Linux 9 moodlebox tty1_
_moodlebox login : pi_
_password : raspberry_

J'ai essayé toutes les combinaisons que je connais mais aucune ne fonctionne. Sauf erreur de ma part la combinaison "pi" / "raspberry" devrait fonctionner mais avec l'image moodlebox je n'arrive pas à me connecter. A la lecture de la documentation Moodle devrait s'exécuter sans avoir besoin de s'identifier. Je ne vois pas d'où peut provenir le problème. Ma configuration est un Raspberry pi3, SDcard de 16Go, MoodleBox 2.1 

J'ai essayé en SSH et le problème est identique (login incorrect).

J'ai cherché sur Internet mais je ne trouve pas le même problème.

Pourriez-vous me donner une piste de recherche ?

Merci pour votre aide.

martignoni, 2018-02-01:
Bien volontiers. Lisez la documentation ici: http://moodlebox.net/fr/help/, en particulier la page https://moodlebox.net/fr/help/noms-dutilisateur-et-mots-de-passe/.

lsteamgeo, 2018-02-01:
Merci beaucoup pour votre réponse. 

Tout fonctionne parfaitement !

closed 2018-02-01 by lsteamgeo