moodlebox/moodlebox #66: Expose Moodle to local LAN network

spipau, 2018-03-31:
Hi,

first of all, thanks for your great work! I just installed it on my Pi and it works out of the box, amazing 👍 

I am working as a guest lecturer in rural India where they recently broke their local Moodle version again and I am sick of fixing their servers.
I would love to propose them Moodlebox, but all they have is a local network with desktop PCs and therefore only LAN cable connections. Wifi connections are not an option. I looked into your documentation and googled a bit, but I couldn't find an answer to my questions:

1. Is it possible to connect my Moodlebox via a LAN cable and make it accessible to desktop PCs via moodlebox.home? (I tried it, out of the box it doesn't work)
2. Does it make sense to implement a new feature to reverse the network connection access point? 

So basically, I would like to make a switch as an admin to deactivate the WIFI, connect my Moodle box to the local LAN and activate "access via LAN mode" so all other PCs in the local network can access Moodle.

Thanks for your time and efforts!
Paul

ralf-krause, 2018-04-01:
Hi Paul, 
the ethernet lan port gets its ip address from the network dhcp server. In small networks this will be the internet router but in some greater networks there will be a separate server for this. If you can get a fixed ip address for the MoodleBox and you can add the domain moodlebox.home to your dns server then you can access the MoodleBox from the ethernet lan. 
To connect the MoodleBox with the internet via the ethernet lan port it should not provide dhcp ip addresses via ethernet. There will be a conflict when more than only one dhcp server is working in a network.
Ralf

ralf-krause, 2018-04-01:
Please look at https://moodlebox.net/en/help/how-to-access-the-moodlebox-via-ethernet/

In the german version I added a configuration to the hosts file. This must be done on every computer which should connect via ethernet. https://moodlebox.net/de/help/zugang-ueber-ethernet/

martignoni, 2018-04-01:
To your questions (to complete what @ralf-krause already wrote):
 
1. As a matter of fact, access to MoodleBox via Ethernet **does work out of the box** via its IP address (e.g. 192.168.1.xxx), distributed by your local DHCP router. However, the Moodle server of the MoodleBox wants to be called exclusively by its address **moodlebox.home**. So you have to configure your local router to give your MoodleBox this name (moodlebox.home). As [described in the documentation](https://moodlebox.net/en/help/how-to-access-the-moodlebox-via-ethernet/), this requires a DNS server on your Ethernet network (e.g. on your router).
To better understand the network infrastructure around the MoodleBox, you may want to look at [this description of the network topology](https://moodlebox.net/en/help/moodlebox-network-topology/).

2. I don't understand what you mean. Please elaborate.

martignoni, 2018-04-01:
> So basically, I would like to make a switch as an admin to deactivate the WIFI, connect my Moodle box to the local LAN and activate "access via LAN mode" so all other PCs in the local network can access Moodle.

Just an idea: instead of deactivate the WiFi, you may change the WiFi password to some secret string, so that your clients cannot access it.

Out of curiosity: why are WiFi not an option?

ralf-krause, 2018-04-01:
Hi Nicolas, perhaps Paul can‘t use wifi because the computers don‘t have wifi. In this case he will have no option for wifi connections. In my workshops I found some schools with old computers without any network or internet. If those computers are standing in a room together you can connect them via ethernet and may start with a MoodleBox. Ralf

martignoni, 2018-04-01:
Maybe. Let Paul tell us :-)

spipau, 2018-04-02:
Hi,

wow, thanks for all the responses and help from your side! (und danke et merci für die deutsche Beschreibung 😉)

I am not a server expert, since I am working more in the Android/iOS sector. I played around a bit in my local network, but I can't get it running. Which is not a big deal, since I am currently living in Paris and I only have a crappy SFR Router Box available:

I can give the Moodlebox a static IP, but I have no access to the DNS server. Also nothing happens when I change my local hosts file on my MacBook (+restart). 

I will try to get a better set-up, but that might take some time. Or I give it a try during my next India trip. The good thing now is that, thanks to you, I know that it is possible and I know how to do it!

> perhaps Paul can‘t use wifi because the computers don‘t have wifi. 

Exactly, this is the problem. They only have desktop computers with no wifi. So I would keep the Wifi running, since some of the students have mobile phones, but a LAN connection is key to really use Moodle during a lecture.

Thank you, merci und Danke! I guess you can close the ticket 👍 
Paul

martignoni, 2018-04-03:
Thanks for the feedback, and good luck with the MoodleBox.

PS. Didn't thought there were still around so old PCs as not to have WiFi hardware builtin. Sorry.

closed 2018-04-03 by martignoni

spipau, 2018-04-03:
Not an issue at all! Thank you very much for your efforts and help!