moodlebox/moodlebox #78: Update Raspbian 4.14.50-v7+ #1122

ralf-krause, 2018-06-29:
Hi Nicolas,
I updated my MoodleBox to Raspbian 4.14.50
`Linux moodlebox -v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l`

Everything seems to be okay with the MoodleBox.
Ralf

[image]
[image]

martignoni, 2018-06-29:
Thanks. Precisely: did you do a kernel update?

ralf-krause, 2018-06-29:
My MoodleBox uses the image MoodleBox 2.4.0 and I made a Raspbian update. 
`sudo apt-get update && sudo apt-get upgrade -y`

martignoni, 2018-06-29:
Thanks for the info. It's a kernel update.

I'll close this and open a new issue about upgrading to the next Raspbian image (but after the MoodleMoot! Too much work now :-)).

martignoni, 2018-06-29:
See #79.

closed 2018-06-29 by martignoni