moodlebox/moodlebox #85: Remove display of the Change password link from phpMyAdmin GUI

martignoni, 2018-08-12:
This feature will prevent users to change main password via Change password link in phpMyAdmin, which breaks MoodleBox functionality. 

This **password can and should be changed** using [MoodleBox plugin GUI](url
https://github.com/moodlebox/moodle-tool_moodlebox) within Moodle.

Documentation issue related: 
https://github.com/moodlebox/moodlebox.net/issues/17.

martignoni, 2018-08-12:
See 
https://docs.phpmyadmin.net/en/latest/config.html#cfg_ShowChgPassword for implementation.

frankkoch, 2018-08-12:
Hello Nicolas

Thank you for the additional security. I installed this plugin and now I realize that access via putty (see https://moodlebox.net/de/help/zugang-zum-betriebssystem/) doesn’t work. I’m not sure whether this is related to the upgrade. The error I’m getting is
[cid:image001.png@01D4325E.35F26660]

However, I can still connect via command line ssh.

Kind regards,
Frank

moodlebox/moodlebox.net#17<https://github.com/moodlebox/moodlebox.net/issues/17>.

martignoni, 2018-08-13:
@frankkoch: I'm afraid I don't understand your comment. The feature in this issue is not yet implemented in the current MoodleBox image, and will only be present in the next image of the MoodleBox.

Is your problem really related to this very issue?

frankkoch, 2018-08-13:
Hi Nicolas, under Windows I want to use Putty to connect to the Moodlebox Linux . I followed you manual at https://moodlebox.net/de/help/zugang-zum-betriebssystem/. But instead of asking for a password I'm getting the sent error. However, when I use a Windows commend prompt, I can connect via SSH.

Do I understand right that the feature described at https://moodlebox.net/de/help/zugang-zum-betriebssystem/ will become available in a next version of the Moodlebox? I'm still on v2.4.1.

martignoni, 2018-08-13:
In fact this has nothing to do with the present issue. I contact you via mail, because this is off-topic here.

martignoni, 2018-08-14:
Closing as fixed.

closed 2018-08-14 by martignoni