moodlebox/moodlebox #86: Extending MoodleBox via Wifi Antennae. How?

popeye6554, 2018-08-13:
I want to extend the range of the wifi signal so I tried connecting an access point, but its unable to resolve the DNS. 

I'm using a Nano Station Loco M2 airMax access point. It works with my RPi with DHCP configured, but that was before I tried the MoodleBox. I'm so happy the moodlebox works, but the next step for me is to configure it so that my airMax works...

Any ideas/suggestions?

martignoni, 2018-08-13:
Could you provide a sketch of your intended wireless setup, so that I get a better idea of what you want to do? Much needed to be able to help you.

popeye6554, 2018-08-13:
Hi, 

This is the intended setup that I want: 
Client Devices (phones, laptops, tablets) --- > Nano Station Loco M2 airMax ----> switch ----> RaspberryPi

*The switch is optional. 

Currently, the moodlebox is working out of the box like this:
Client Devices via wifi (phones, laptops, tablets) ----> MoodleBox 

What I want to do is be able to extend the range of the signal as this is a very important step. I've tried to connect the MoodleBox via ethernet into my switch, but from my reading on this page it would seem I would need to have a server running to configure DNS https://moodlebox.net/en/help/how-to-access-the-moodlebox-via-ethernet/ 

So based on my previous experience of knowing that the RPi can work with the NanoStation antennae, I'm trying to see what can be done. 

NB: My RaspberryPi ethernet port was damaged sometime ago so I use a USB To Ethernet Dongle with it to connect the RaspberryPi to the Switch.

What I've tried: I've tried changing the Gateway for the switch to 10.0.0.1 and the IP address for the switch has been set to static and given 10.0.0.5, however I'm getting a DHCP configuration error when I try to connect to the signal from the NanoStation.

martignoni, 2018-09-20:
No feedback received. Closing.

closed 2018-09-20 by martignoni