moodlebox/moodlebox #90: URL shortener for the MoodleBox

ralf-krause, 2018-09-28:
Hi Nicolas,
what do you think about an url shortener for the MoodleBox? 

I had a teachers workshop for teachers with a very interesting setting. The teachers should find a hidden MoodleBox in the wood. The used the smartphone app OsmAnd with an OpenStreetMap to find a MoodleBox by its gps coordinates. The MoodleBox had no wifi passwords everybody could connect without knowing a password.

To open a Moodle course firstly I thought about using a qr code. But I should use something digital .... a digital idea .... something like an iBeacon. Okay ... I tested the calliope mini as a beacon ... please look at https://ckblog2016.wordpress.com/2017/08/11/calliope-mini-als-ble-beacon/

The problem was that the beacon allows only 17 characters for the url. Every url expect the start page of the MoodleBox has more than 17 characters.  https://makecode.microbit.org/reference/bluetooth/advertise-url

In our offline setting I could not use a normal url shortener from the Internet. There was no connection to the internet. So I added some hard coded short links http://moodlebox/x/t1 for the team 1 and http://moodlebox/x/t2 for the team 2. I did this with a folder x and two subfolders t1 and t2. Inside t1 i loaded an index.php with an http location reload to http://moodlebox.home/course/view.php?id=7&section=1

I am thinking about a local url shortener because no teacher will get an idea how to do this. If there would be an url shortener we would get very easy links like http://mb/x/t1. The url shortener should have a web gui to create and to edit the short links. 

Best regards, Ralf

ralf-krause, 2018-09-29:
Hi Nicolas,
what should I do to add this url shortener to the MoodleBox?
https://github.com/mathiasbynens/php-url-shortener
Ralf

martignoni, 2018-09-29:
> What should I do to add this url shortener to the MoodleBox?
> https://github.com/mathiasbynens/php-url-shortener

You should follow the instruction given by the author, since the MoodleBox is a standard UNIX server. Please do ask him if you want support about this, because I don't know this tool.

martignoni, 2018-09-29:
I'm afraid I won't implement this, because I want to keep the MoodleBox simple, following the Unix philosophy principle: [Do One Thing And Do It Well](https://en.wikipedia.org/wiki/Unix_philosophy#Do_One_Thing_and_Do_It_Well).

Two more reasons to this:
1. Keep it simple and straightforward for people to use and for me to maintain.
1. The more you add, the more complex the support becomes.

martignoni, 2018-09-30:
For QR codes, I myself use: http://goqr.me/. Re url-shortener: it would be bold that I give support or advice on tools that I absolutely don't know. You would have better luck asking the developer.

ralf-krause, 2018-10-03:
Hi Nicolas,
I wanted to get short urls to access to the Moodlebox with some bluetooth beacons. A beacon allows to have an url with 16 characters only. So I thought about short urls like http://mb/u/?i=c1 or short urls like http://mb/u/c1

Now I have found three options for this. Please look at my documentation in the attached file. Sorry, the description is in german language but you will find all needed informations and file content.

Best regards, Ralf

[url-shortener.txt](https://github.com/moodlebox/moodlebox/files/2443492/url-shortener.txt)

ralf-krause, 2018-10-07:
Hi Nicolas, 
I got it! The url shortener YOURLS is running on my MoodleBox.

First I should tell why I could not use an url shortener bit.ly or goo.gl or tiny url.com ... my project runs offline on the MoodleBox and I couldn't use an online url shortener. I needed to get very short urls on my MoodleBox because iBeacons should be used for linking to some Moodlebox content ... and iBeacons only allow urls with 16 chars. 

So to get a very short base url I edited the file `/etc/hosts` on my Moodlebox and added `mb` to the last line. This shows the start page with the short url `http://mb/` and a course with `http://mb/course/view.php?id=2` ... but the url to any course is still too long for the iBeacon.
```
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

10.0.0.1        moodlebox.home  moodlebox  mb
```

I installed the files from `http://yourls.org` into the folder `/var/www/yourls/` and added a new database `yourls` with the collation `utf8_general_ci` via phpMyAdmin. I copied the file `user/config-sample.php` to `user/config.php` and edited the database settings.

```
/** MySQL database settings */
define( 'YOURLS_DB_USER', 'moodlebox' );

/** MySQL database password */
define( 'YOURLS_DB_PASS', 'Moodlebox4$' );

/** The name of the database for YOURLS */
define( 'YOURLS_DB_NAME', 'yourls' );

/** MySQL hostname.
 ** If using a non standard port, specify it like 'hostname:port', 
 ** eg. 'localhost:9999' or '127.0.0.1:666' */
define( 'YOURLS_DB_HOST', 'localhost' );

/** MySQL tables prefix */
define( 'YOURLS_DB_PREFIX', 'yourls_' );

/** MySQL database engine */
define( 'YOURLS_DB_DRIVER', 'mysqli' );

/** YOURLS installation URL -- all lowercase and with no trailing slash.
 ** If you define it to "http://sho.rt", don't use "http://www.sho.rt" in your browser (and vice-versa) */
define( 'YOURLS_SITE', 'http://mb/u' );

```

In the Moodle folder I added a link to the url shortener ... the url will be `http://mb/u` ... it's very short  :-)
```
cd /var/www/moodle
sudo ln -s /var/www/yourls u
```

To rewrite the short urls in the web server nginx I had edit the file `/etc/nginx/sites-available/default` ... I added some lines at the bottom of the file and rebooted the MoodleBox.
```
        ### url shortener
        location /u/ {
                try_files $uri $uri/ /u/yourls-loader.php$is_args$args;
        }
```

This was all ... the database installed automatically on the first start of the admin page `http://mb/u/admin`. 

You will see three predefined short urls in the admin interface and you can add your own urls. I started with `http://moodlebox.home/course/view.php?id=2` and the short url `http://mb/u/c2/` ... very short ... very nice! The next thing I tested was `http://moodlebox.home/course/view.php?id=7&section=2` and the short url `http://mb/u/c7s2/`.

   
Some links from my project
- [https://ckblog2016.wordpress.com/2017/08/11/calliope-mini-als-ble-beacon/](https://ckblog2016.wordpress.com/2017/08/11/calliope-mini-als-ble-beacon/)
- [https://makecode.microbit.org/v1/reference/bluetooth/advertise-url](https://makecode.microbit.org/v1/reference/bluetooth/advertise-url)
- [http://yourls.org](http://yourls.org)

[image]

martignoni, 2018-10-12:
Thanks. Closing now.

closed 2018-10-12 by martignoni