moodlebox/moodlebox #142: Moodle Plugin installation

martignoni, 2019-07-14:


martignoni, 2019-07-14:
This behavior cannot be consistently reproduced.

martignoni, 2019-07-14:
Use following shell command if you experience this.
```
sudo systemctl reload php7.3-fpm
```

martignoni, 2019-07-14:
Reports indicate that this is independent of the RPi model: has occurred on 3B+, 4B (1GB), 4B (2GB).

martignoni, 2019-07-14:
Maybe this: https://bugs.php.net/bug.php?id=77955

ralf-krause, 2019-07-15:
The same problem also happened with the new MoodleBox image and a Raspberry Pi 3B (not 3B+).
I did not install a plugin. I logged in and opened the website administration properly. After this I could not open the MoodleBox plugin anymore. Yes, I reload the php7.3-fpm and then it worked.

ralf-krause, 2019-07-16:
Today I got the same problem when I tried to change a plugin setting. 
I opened the page for the general feedback setting http://moodlebox.home/admin/settings.php?section=modsettingfeedback and set "allow full anonymous" to yes. When I saved the changes I got the problem. After reloading php7.3-fpm I got everything working again.

ralf-krause, 2019-08-11:
Hi Nicolas,
today I updated the Raspbian to the newest version. PHP seems to be PHP 7.3.4 as it was before. But a lot of other packages were updated. After reboot I tested to install some plugins without any problem. It could be the the reason for this issue is fixed now. I was using a Raspberry Pi 4B for this test.
Ralf

Linux moodlebox 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l

martignoni, 2019-08-11:
Thanks for the report. Same here, unable to reproduce since about 2016-07-15.

Let's however keep this open for some days, to be sure that this is gone.

martignoni, 2019-10-30:
Seems to occur from time to time even with 2019-09-26 Raspbian release.

artyuum, 2019-12-16:
Same on WSL (php7.3-fpm).

martignoni, 2020-02-11:
Hopefully fixed with 2020-02-05 Raspbian release? Will see...

martignoni, 2020-02-12:
As a temporary workaround, we could add this to root `crontab`:
```
*/5 * * * * systemctl restart php7.3-fpm.service
```

ralf-krause, 2020-02-21:
Hi Nicolas,
there is an update php7.3-cli (7.3.14-1~deb10u1) for Raspbian. I can't say if this update is already part of the 2020-02-05 Raspbian release.
Ralf

martignoni, 2020-02-21:
Installed version is `php7.3-cli/now 7.3.11-1~deb10u1`. But what interests us here is php-fpm, i.e. `php7.3-fpm/now 7.3.11-1~deb10u1`.

ralf-krause, 2020-02-21:
Looking at http://moodlebox.home/admin/phpinfo.php I get the same version `PHP Version 7.3.14-1~deb10u1`.

martignoni, 2020-02-21:
This is because you updated your installation. I've done it too and get the same as you.

ralf-krause, 2020-07-19:
Could you see if this issue is repaired in Raspberry Pi OS 64 bit?

martignoni, 2020-07-20:
This will have to wait a release of RPi OS 64bits **lite**, or at least a Release Candidate. Anyway, I'm carefully monitoring this.

martignoni, 2020-07-23:
This indicates another workaround: https://kadai.com.mx/sigsegv-en-php-7-3-soluciones-al-problema/.

martignoni, 2020-07-24:
PHP configuration is not optimal and leads to these PHP outages. Tweaking PHP settings fixes the problem.

closed 2020-07-24 by martignoni