moodlebox/moodlebox #145: Update Raspbian to 4.19.66-v7l+ #1253

ralf-krause, 2019-08-20:
Hi Nicolas,
I updated Raspbian to 4.19.66-v7l+ #1253 on my Moodlebox without any problem.
`Linux moodlebox 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux`
The Raspbian download image 4.19 is still an older one dated on 2019-07-10.
Ralf

martignoni, 2019-08-20:
This is normal. I don't release new images every time Debian (and Raspbian) packages are updated; this would be a nightmare.

New MoodleBox images are released following the roadmap from the [project milestones](https://github.com/moodlebox/moodlebox/milestones). When I release a new image, it is always up-to-date, even between Raspbian image releases. This means I update the distro packages in the normal release process.

Between releases, it's very easy as you know for tech savvy people to update their packages, following the [official doc](https://moodlebox.net/en/help/how-to-upgrade-the-moodlebox/). Would you suggest another process?

ralf-krause, 2019-08-20:
I thought that this information could be important becausew the small update 4.19.58-v7l+ #1245 could remove the problems with php7.3-fpm. I did not get any problem since I updated.
See https://github.com/moodlebox/moodlebox/issues/142#issuecomment-511875510

martignoni, 2019-08-20:
This can be a good reason to release a new image ASAP. I'll consider it.

martignoni, 2019-08-23:
Building new image now. Closing.

closed 2019-08-23 by martignoni