moodlebox/moodlebox #153: Update to version 4.3.0 of Nodogsplash

martignoni, 2019-10-08:
See https://github.com/nodogsplash/nodogsplash/releases/tag/v4.3.0.

martignoni, 2019-10-08:
Blocked by https://github.com/nodogsplash/nodogsplash/issues/447.

martignoni, 2019-10-09:
We'll skip to https://github.com/nodogsplash/nodogsplash/releases/tag/v4.3.1.

martignoni, 2019-10-10:
We'll skip to https://github.com/nodogsplash/nodogsplash/releases/tag/v4.3.2.

martignoni, 2019-10-10:
Test OK, closing.

closed 2019-10-10 by martignoni

bluewavenet, 2019-10-20:
@martignoni 
We found obscure an issue with NDS v4.3.0/1/2 causing it to lock up or crash under high load or with high CPD rates (Windows 10 Desktop CPD has been seen polling the captive portal many times per second when the client is connected but not authenticated for example).
The cause was a coding error resulting in a race condition in libmicrohttpd with a resultant memory corruption.

NDS version 4.3.3 is now released fixing the issue. 

You quite likely will not have seen the issue with your application, but never the less I would suggest upgrading to this new version. Apologies!

bluewavenet, 2019-10-20:
https://github.com/nodogsplash/nodogsplash/releases/tag/v4.3.3

martignoni, 2019-10-20:
@bluewavenet Thanks for the heads up. Update is planned: https://github.com/moodlebox/moodlebox/issues/155.