moodlebox/moodlebox #171: Investigate Raspberry Pi OS 64 bit

martignoni, 2020-05-28:
Raspberry Pi OS 64 bit was just announced: https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370.

Let's try to use it for the MoodleBox!

martignoni, 2020-06-07:
Just tested Raspberry Pi OS 64 bit image.

It works well 😄!

martignoni, 2020-08-25:
RPI OS 64 bits Lite image was released shortly: https://downloads.raspberrypi.org/raspios_lite_arm64/images/.

martignoni, 2020-10-28:
Will be integrated when out of beta.

martignoni, 2021-04-11:
[New version of Raspberry Pi OS Lite 64 bit released](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-04-09/).

v3ss0n, 2021-09-16:
Currently it is 32bit? Any updates on going 64 bit?

martignoni, 2021-09-17:
See https://discuss.moodlebox.net/d/260-plans-for-moodlebox-bullseye-and-64-bit/.

martignoni, 2021-11-08:
Raspberry Pi OS with Debian Bullseye released is only 32bit. We won't release a 64bit version of MoodleBox until Raspberry Pi OS64bit is officially released.

v3ss0n, 2021-11-18:
That make sense , i was trying to run  MoodleBox on qemu with  64bit ARM RPI4 emulation

martignoni, 2022-02-02:
Raspberry Pi OS 64-bit is now out of beta 😄 !

martignoni, 2022-04-18:
Works as expecting. Closing.

closed 2022-04-18 by martignoni