moodlebox/moodlebox #176: Wifi disabled when swapping SD card to other devices

martignoni, 2020-06-14:
As noted in several discussions, Wifi can be disabled when swapping SD card to other devices.

Command `sudo rfkill unblock wlan` fixes the problem.

See:
- https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275637&p=1669869
- https://discuss.moodlebox.net/d/157-new-moodlebox-installation-not-providing-connectivity

martignoni, 2020-06-14:
Workaround in developement.

ralf-krause, 2020-06-16:
Hi Nicolas,
I got the wifi problem with a fresh installation of MoodleBox 3.7.0 and a Raspberry Pi 3B+.
Yes, the terminal command `sudo rfkill unblock wlan` started the wifi and I could connect.
Ralf

martignoni, 2020-06-16:
Unfortunately, the problem is not fixed yet. Reopening.

scara, 2020-06-16:
Hi @martignoni,
why not appending that comand in [`rc.local`](https://www.raspberrypi.org/documentation/linux/usage/rc-local.md)?

Wondering if the root cause could be _Network Manager_, i.e. https://askubuntu.com/a/1034237 or just try https://www.reddit.com/r/linuxquestions/comments/6arv5l/unblock_wifi_with_rfkill_on_boot/dhh5935/ i.e. sleeping a couple of minutes and than issueing your comand.

HTH,
Matteo

martignoni, 2020-06-16:
Thanks Matteo, I could have indeed make the workaround work by slightly delaying the `@reboot` cron command, but I prefer to fix the problem at its root.

If you're interested, the problem was that RPi OS (and older Raspbian) image choose to explicitely block the wifi by writing two files in the `systemd` hierarchy: one for RPi 3B and the second for RPi 4B. See here: https://github.com/RPi-Distro/pi-gen/blob/08fc0b9a8237c5a9926a0e9d660156a1e838d33a/stage2/02-net-tweaks/01-run.sh#L28-L31.

As I prepared my image on a RPi 4B, my `rfkill` command would only change the bit on the second of these files, so that when the image was used on a RPi 3B, the problem still occured. And as written above, my workaround did not work, because the command was issued too early during the boot process.

I'm pretty confident that this will be fixed now.

scara, 2020-06-16:
TNX @martignoni for the reference!
Really appreciated 👍.

scara, 2020-06-16:
For the record: https://github.com/RPi-Distro/pi-gen/pull/416.

HTH,
Matteo

martignoni, 2020-07-24:
Tested. Works as expected :-). Closing.

closed 2020-07-24 by martignoni