moodlebox/moodlebox #187: Failed to start dnsmasq

heinzmarchl, 2020-11-01:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Describe the bug
A clear and concise description of what the bug is.

### To Reproduce
Complete and precise steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior
A clear and concise description of what you expected to happen.

## Additional context
Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.
- Raspberry Pi model [e.g. RPi 3B+]
- MoodleBox version
- MoodleBox plugin version

## Screenshots
If applicable, add screenshots to help explain your problem.

martignoni, 2020-11-01:
Please complete the form. Without more info, impossible to fix anything.

martignoni, 2020-11-12:
Will be closed in a few days when no details given.

heinzmarchl, 2020-11-12:
rassberry pi 4 8GB
startscreen
Problem to connect with moodlebox from another computer(windows10, and ipad
lan connection ok.
failed  failed to start dnsmasq - a lightweight DHCP and caching DNS server
orginal Netzteil 
[image]
what can I do?
mfg
Heinz Marchl

martignoni, 2020-11-12:
Impossible to know without more context.

To be able to investigate, we need **precise instructions** on how to reproduce the issue, which are still missing. More over, RPi model, MoodleBox image version, etc. must be given.

Please complete the above form, specially the _To reproduce_ section with precise steps and the _Additional context_ section, and, if applicable, what actions you made before the problem occured. Only with such info can we investigate if there's really an issue and try to fix it.

martignoni, 2020-11-12:
Or maybe is this no bug and you just need support? In this case, you're welcome on the [MoodleBox discussion forum](https://discuss.moodlebox.net/).

martignoni, 2020-12-05:
Closing, not a bug.

closed 2020-12-05 by martignoni