moodlebox/moodlebox #195: Raspberry Pi 3 used to support 32 WiFi connections but is now limited to ~10. RPi 3 B+ and RPi 4 are now limited to ~4

martignoni, 2021-03-02:
This unsolved problem is beeing discussed for 2-3 years in several lists and forums. e.g.:
- [MoodleBox](https://discuss.moodlebox.net/d/36-low-number-of-wi-fi-concurrent-connected-users) [own](https://discuss.moodlebox.net/d/134-very-low-amount-of-wifi-clients-can-connect) [discussion](https://discuss.moodlebox.net/d/203-wlan-problem-netzwerk-accesspoint) [forum](https://discuss.moodlebox.net/d/233-probleme-de-connexion-wifi)
- [Raspberry Pi discussion forum](https://github.com/raspberrypi/linux/issues/3010)
- xsce-devel mailing list

A solution seems to [downgrade the WiFi chip firmware](https://github.com/iiab/iiab/issues/823#issuecomment-666547997).

martignoni, 2021-03-02:
See https://github.com/iiab/iiab/pull/2472.

martignoni, 2021-03-02:
See also: https://github.com/raspberrypi/firmware/issues/1403#issuecomment-648316896 and following comments.

martignoni, 2021-03-02:
Peliminary testing shows an improvement :-)

closed 2021-03-03 by martignoni

ralf-krause, 2021-03-04:
This seems to be e very good idea. Thank you very much for this patch. 

What happens when I update the system via "sudo apt-get update && sudo apt-get dist-upgrade -y". 
Is the older firmware in a secure place so the update will do nothing with the wifi firmware?

Ralf

martignoni, 2021-03-04:
> What happens when I update the system via "sudo apt-get update && sudo apt-get dist-upgrade -y".

I must admit that I did not test this for this pre-release version. Will do and take action.

ralf-krause, 2021-03-06:
The new image does not work for me. I get no wifi on a Raspberry Pi 4B with 4GB RAM.
https://github.com/moodlebox/moodlebox/releases/download/v3.10.0-dev/moodlebox-3.10.0-dev.img.xz

ralf-krause, 2021-03-07:
The new image works with a Raspberry Pi 3B+. I could connect 11 devices the the MoodleBox wifi.
https://github.com/moodlebox/moodlebox/releases/download/v3.10.0-dev/moodlebox-3.10.0-dev.img.xz

martignoni, 2021-03-10:
Please follow up here: https://github.com/moodlebox/moodlebox/issues/199.

martignoni, 2021-03-13:
> What happens when I update the system via "sudo apt-get update && sudo apt-get dist-upgrade -y".
> Is the older firmware in a secure place so the update will do nothing with the wifi firmware?

PR #200 should prevent any unwanted firmware replacement.

ralf-krause, 2021-03-13:
Is there a new image which prevents any unwanted firmware replacement? Or should the image work which you published 6 days ago?

ralf-krause, 2021-03-13:
Before I updated
```
moodlebox@moodlebox:~ $ uname -a
Linux moodlebox 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l
moodlebox@moodlebox:~ $ cd /lib/firmware/brcm/
moodlebox@moodlebox:/lib/firmware/brcm $ ls -al brcmfmac43455*
-rw-r--r-- 1 root root 488193 Mar  2 21:06 brcmfmac43455-sdio.bin
-rw-r--r-- 1 root root 631467 Jan 25 11:17 brcmfmac43455-sdio.bin.2021-03-02
-rw-r--r-- 1 root root  14036 Mar  2 21:06 brcmfmac43455-sdio.clm_blob
-rw-r--r-- 1 root root   7163 Jan 25 11:17 brcmfmac43455-sdio.clm_blob.2021-03-02
-rw-r--r-- 1 root root   2074 Jan 25 11:17 brcmfmac43455-sdio.txt
moodlebox@moodlebox:/lib/firmware/brcm $
```

... and after the update
```
moodlebox@moodlebox:~ $ uname -a
Linux moodlebox 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
moodlebox@moodlebox:~ $ cd /lib/firmware/brcm/
moodlebox@moodlebox:/lib/firmware/brcm $ ls -al brcmfmac43455*
-rw-r--r-- 1 root root 488193 Mar  2 21:06 brcmfmac43455-sdio.bin
-rw-r--r-- 1 root root 631467 Jan 25 11:17 brcmfmac43455-sdio.bin.2021-03-02
-rw-r--r-- 1 root root  14036 Mar  2 21:06 brcmfmac43455-sdio.clm_blob
-rw-r--r-- 1 root root   7163 Jan 25 11:17 brcmfmac43455-sdio.clm_blob.2021-03-02
-rw-r--r-- 1 root root   2074 Jan 25 11:17 brcmfmac43455-sdio.txt
moodlebox@moodlebox:/lib/firmware/brcm $
```

martignoni, 2021-03-13:
All released images are available here: https://github.com/moodlebox/moodlebox/releases.

As already stated, v3.10.0-dev is a developement version, not to be used for production. The latest **supported** image is still v3.9.0. The next one will be released shortly.

ralf-krause, 2021-03-14:
After I updated the MoodleBox I no longer got the wifi "MoodleBox", although the firmware for the wifi chip has remained unchanged. I will reinstall the image moodlebox-3.10.0-dev again.

martignoni, 2021-04-20:
https://blog.quarkslab.com/reverse-engineering-broadcom-wireless-chipsets.html

commit 1ee4d104 (2021-03-03): Add role to replace wifi firmware (#196)

- Fixes #195.