moodlebox/moodlebox #197: Configure the server paths

ralf-krause, 2021-03-08:
Hi Nicolas,
it should be a good idea to configure the server paths in Moodle. 
https://docs.moodle.org/310/en/System_paths

The paths for php, du, gs and python3 are easy to find. aspell and dot are not installed on the MoodleBox.

If you insert the php path then the admin is able to start all scheduled tasks without waiting. http://moodlebox.home/admin/tool/task/scheduledtasks.php

If you insert the du path then folders with a lot of files should be shown much faster.

The gs path is already set so that Ghostscript works for generating and annotating pdf documents.
https://docs.moodle.org/310/en/Assignment_settings#Feedback_plugins

The python3 path should be set for Moodle Analytics. https://docs.moodle.org/310/en/Analytics_settings#Configure_learning_analytics_settings

Best regards, Ralf

[image]

martignoni, 2021-03-08:
Good idea indeed.

martignoni, 2021-03-10:
TODO: Documentation of default Moodle configuration.

martignoni, 2021-03-10:
Documentation updated. See https://github.com/moodlebox/moodlebox.net/commit/fbeb9698f90578e4d911f28d72b1957c4b306a9d.

closed 2021-03-10 by martignoni