moodlebox/moodlebox #199: No Wi-Fi on RPi 4B with image 3.10.0-dev

martignoni, 2021-03-10:
## Describe the bug
Wi-Fi doesn't work on Raspberry Pi 4B (4GB RAM). See https://discuss.moodlebox.net/d/36-low-number-of-wi-fi-concurrent-connected-users/61. Same user (@ralf-krause) reports a working image on RPi 3B+.

### To Reproduce
Complete and precise steps to reproduce the behavior:
1. Download [development version 3.10.0-dev](https://github.com/moodlebox/moodlebox/releases/download/v3.10.0-dev/moodlebox-3.10.0-dev.img.gz). 
2. Copy the image
3. Startup RPi
4. Observe that no `moodlebox`Wi-Fi SSID is published.

## Expected behavior
A Wi-Fi SSID `moodlebox` should be aired.

martignoni, 2021-03-10:
@ralf-krause: Could you please the output of the following commands?
```
cat /var/log/syslog | grep hostapd
```
and
```
cat /var/log/syslog | grep dnsmasq
```

ralf-krause, 2021-03-10:
Hi Nicolas, 
I switched the wifi regulatory country to DE. Also I changed the wifi channel from 11 to 10 this afternoon because there is another access point with wifi channel 13. But I do not get a better result. 
Ralf

```
Last login: Wed Mar 10 15:01:45 2021 from 192.168.1.196
moodlebox@moodlebox:~ $ cat /var/log/syslog | grep hostapd
Mar 10 06:31:27 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: associated
Mar 10 06:31:27 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 RADIUS: starting accounting session 8EC3B5E1A3B51C14
Mar 10 06:31:27 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 WPA: pairwise key handshake completed (RSN)
Mar 10 06:48:08 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 06:48:08 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 06:52:12 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: associated
Mar 10 06:52:12 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 RADIUS: starting accounting session D3AD9D2B988DE650
Mar 10 06:52:12 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 WPA: pairwise key handshake completed (RSN)
Mar 10 07:02:33 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 07:02:33 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 07:07:17 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: associated
Mar 10 07:07:17 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 RADIUS: starting accounting session C50441C775060771
Mar 10 07:07:17 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 WPA: pairwise key handshake completed (RSN)
Mar 10 07:58:30 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 07:58:37 moodlebox hostapd: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 13:53:45 moodlebox systemd[1]: hostapd.service: Succeeded.
Mar 10 13:53:45 moodlebox hostapd[12011]: Configuration file: /etc/hostapd/hostapd.conf
Mar 10 13:53:45 moodlebox hostapd[12011]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mar 10 13:53:47 moodlebox systemd[1]: hostapd.service: Succeeded.
Mar 10 13:53:47 moodlebox hostapd[12030]: Configuration file: /etc/hostapd/hostapd.conf
Mar 10 13:53:47 moodlebox hostapd[12030]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mar 10 13:53:47 moodlebox hostapd[12030]: wlan0: Could not connect to kernel driver
Mar 10 13:53:47 moodlebox hostapd[12030]: Using interface wlan0 with hwaddr dc:a6:32:0f:a2:b0 and ssid "MoodleBox"
Mar 10 13:53:47 moodlebox hostapd[12030]: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mar 10 13:53:47 moodlebox hostapd[12030]: wlan0: AP-ENABLED
Mar 10 15:09:43 moodlebox hostapd[599]: Configuration file: /etc/hostapd/hostapd.conf
Mar 10 15:09:43 moodlebox hostapd[599]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
moodlebox@moodlebox:~ $ 
```

```
moodlebox@moodlebox:~ $ cat /var/log/syslog | grep dnsmasq
Mar 10 15:09:43 moodlebox systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Mar 10 15:09:43 moodlebox dnsmasq[601]: dnsmasq: syntax check OK.
Mar 10 15:09:43 moodlebox dnsmasq[610]: dnsmasq: failed to create listening socket for 10.0.0.1: Cannot assign requested address
Mar 10 15:09:43 moodlebox dnsmasq[610]: failed to create listening socket for 10.0.0.1: Cannot assign requested address
Mar 10 15:09:43 moodlebox dnsmasq[610]: FAILED to start up
Mar 10 15:09:43 moodlebox systemd[1]: dnsmasq.service: Control process exited, code=exited, status=2/INVALIDARGUMENT
Mar 10 15:09:43 moodlebox systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Mar 10 15:09:43 moodlebox systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Mar 10 15:09:48 moodlebox systemd[1]: dnsmasq.service: Service RestartSec=5s expired, scheduling restart.
Mar 10 15:09:48 moodlebox systemd[1]: dnsmasq.service: Scheduled restart job, restart counter is at 1.
Mar 10 15:09:48 moodlebox systemd[1]: Stopped dnsmasq - A lightweight DHCP and caching DNS server.
Mar 10 15:09:48 moodlebox systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Mar 10 15:09:48 moodlebox dnsmasq[820]: dnsmasq: syntax check OK.
Mar 10 15:09:49 moodlebox dnsmasq[829]: Too few arguments.
Mar 10 15:09:49 moodlebox systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
moodlebox@moodlebox:~ $ 
```

martignoni, 2021-03-10:
Thanks. Could you please post now the output of these commands?
```
sudo systemctl status hostapd.service
```
and 
```
sudo systemctl status dnsmasq.service
```
and
```
iwconfig
```

ralf-krause, 2021-03-10:
```
moodlebox@moodlebox:~ $ sudo systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-03-10 17:20:51 CET; 1h 25min ago
  Process: 587 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=0/SUCCESS)
 Main PID: 602 (hostapd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hostapd.service
           └─602 /usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Mar 10 18:02:43 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 18:02:43 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 18:03:33 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: associated
Mar 10 18:03:33 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 RADIUS: starting accounting session 09EAE2B50B348DB7
Mar 10 18:03:33 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 WPA: pairwise key handshake completed (RSN)
Mar 10 18:43:33 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated due to inactivity
Mar 10 18:43:34 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mar 10 18:44:05 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 18:44:05 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
Mar 10 18:44:05 moodlebox hostapd[602]: wlan0: STA 02:82:77:a6:64:69 IEEE 802.11: disassociated
moodlebox@moodlebox:~ $ 
```

```
moodlebox@moodlebox:~ $ sudo systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-03-10 17:21:02 CET; 1h 25min ago
  Process: 815 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
  Process: 816 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 824 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
 Main PID: 823 (dnsmasq)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/dnsmasq.service
           └─823 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --l

Mar 10 17:21:02 moodlebox systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Mar 10 17:21:02 moodlebox dnsmasq[815]: dnsmasq: syntax check OK.
Mar 10 17:21:02 moodlebox dnsmasq[824]: Too few arguments.
Mar 10 17:21:02 moodlebox systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.

moodlebox@moodlebox:~ $ 
```

```
moodlebox@moodlebox:~ $ iw dev wlan0 info
Interface wlan0
	ifindex 3
	wdev 0x1
	addr dc:a6:32:0f:a2:b0
	ssid MoodleBox
	type AP
	wiphy 0
	channel 10 (2457 MHz), width: 20 MHz, center1: 2457 MHz
	txpower 31.00 dBm
moodlebox@moodlebox:~ $ 
```

ralf-krause, 2021-03-10:
Now my mobile devices are showing the wifi "MoodleBox" but can't connect ... and a moment later the wifi has gone.

ralf-krause, 2021-03-11:
I don‘t know what happened .... I changed nothing. The MoodleBox is running for more than 16 hours. I  am connected to the wifi „MoodleBox“ since a few minutes with two devices. Yesterday the wifi „MoodleBox“ was not shown on the same devices.

martignoni, 2021-03-11:
Thanks for the feedback. Maybe due to the local environment?

I'll leave this still open a few days, just in case.

ralf-krause, 2021-03-11:
And suddenly the wifi „MoodleBox“ stops working for about 10 minutes and comes back again.

ralf-krause, 2021-03-11:
I can‘t say if this would be important. I found that the MoodleBox dashboard shows the regulatory country is DE and the Android wifi scanner teils me CH. Could there be a problem?
Ralf

[image]

[image]

ralf-krause, 2021-03-11:
Without any digital communication in the wifi "MoodleBox" and without moving around any digital devices the iwi goes away for 2 or 3 minutes and comes back again. The scanner changes the colour from green to yellow. While I 'm writing this text the wifi has got away after 5 minutes again. 2 minutes later the wifi scanner shows the wifi for 4 seconds ... it was too fast to take a photo.

[image]

[image]

ralf-krause, 2021-03-11:
Now I was fast enough ... two peaks from the MoodleBox and then no signal again. The numbers at the bottom are showing the time ... the wifi is scanned every 2 seconds.

[image]

martignoni, 2021-03-15:
I still cannot reproduce this.

ralf-krause, 2021-03-29:
I get the same result with the final image version 3.10.0 .... no wifi with a RPI 4B.

My both RPi 4B (4GB RAM) don't provide a stable wifi signal but only some peaks in the diagram from the Android wifi analyser. I put the micro sd card into a RPi 3B+ and I got a strong and stable wifi signal.

martignoni, 2021-03-29:
This is expected, as there was no modification (for this topic) in the final image.

Unfortunately I did not receive other similar feedbacks from anyone and still cannot reproduce :-(

jvonau, 2021-04-20:
> I can‘t say if this would be important. I found that the MoodleBox dashboard shows the regulatory country is DE and the Android wifi scanner teils me CH. Could there be a problem?

Question for all, was raspi-config used to set the country code at some point either before creation or after installation? The country code should be set in hostapd.conf and wpa_supplicant.conf to the same value in both places.

jvonau, 2021-04-20:
Downloaded the latest image, extracted, loop mounted the image did chroot into the image and this is what I found:
> root@box:/# cat /etc/wpa_supplicant/wpa_supplicant.conf 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=CH
root@box:/#

jvonau, 2021-04-20:
https://github.com/moodlebox/moodlebox/blob/45ce42cb63e9351411c8e7cd742ac0d3487350e4/default.config.yml#L45

martignoni, 2021-04-20:
Indeed. Source: https://github.com/moodlebox/moodlebox/blob/45ce42cb63e9351411c8e7cd742ac0d3487350e4/roles/wifi/tasks/main.yml#L7

jvonau, 2021-04-20:
easy fix is the have the dashboard write the save value in both locations when the user changes the country, but the way RaspOS is plumbed(I can expand on that if you want), a reboot is needed.

martignoni, 2021-04-20:
Right. But IMHO no need to fix it, since MoodleBox doesn't use WPA Supplicant (it isn't used as a WiFi client at all).

Or am I missing something?

jvonau, 2021-04-20:
You are missing something, wpa_supplicant is started anyway by dhcpcd via /lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant. One could delete the file or use "nohook 10-wpa_supplicant" in the dhcpcd.conf file. What you are seeing is a fight for control of the device. What gets weird is https://wireless.wiki.kernel.org/en/developers/regulatory/crda mentions iw and wpa_supplicant as the means to alter the country code, its been awhile I can't recall if hostapd can alter the country code when starting and wpa_supplicant is not running.

martignoni, 2021-04-20:
Thanks. See https://github.com/moodlebox/moodlebox/issues/204.

ralf-krause, 2021-05-18:
Hello Nicolas,
I got the same problem with the new MoodleBox 3.11.0 image. I copied the new image and tested it without any configuration with a Raspberry Pi 4B (4GB RAM) and a Raspberry Pi 3B+.

The Raspberry Pi 4B does not provide a stable wifi so you can't use the MoodleBox. You get a wifi signal for a few seconds and then no signal for some more seconds or minutes. You can take the same pictures from my android wifianalyser I send in March 2021.

The Raspberry Pi 3B+ works fine. The wifi signal is stable and without any breakdown. I connected a few mobile devices to the MoodleBox, logged in as the moodlebox user and could use the site administration.

I hope you will find a solution for this issue. Can you please tell me how to get the original wifi driver running? Is it still installed in the image or do I need to copy it from another installation?

Best regards, Ralf

martignoni, 2021-05-18:
> I hope you will find a solution for this issue.

I'd like to, but you seem to be the unique person to experience this issue. I cannot reproduce it either, so I'm completely in the blue and absolutely don't know what to do about it.

> Can you please tell me how to get the original wifi driver running? Is it still installed in the image or do I need to copy it from another installation?

Just checking tonight and get you a procedure.

ralf-krause, 2021-05-18:
Hello Nicolas,
do you have an overview how many MoodleBox installations exist on a Raspberry Pi 4B? 

I'm mostly using the Raspberry Pi 3B+ in my workshops because the Raspberry Pi 3B+ can be powered with a normal PowerBank. The Raspberry Pi 4B is used on my desktop and I'm connected via lan to the Raspberry Pi 4B.

Ralf

martignoni, 2021-05-18:
I've absolutely no idea of such numbers (either on 3A, 3B, 3B+ or 4B).

martignoni, 2021-06-01:
Another report also with RPi 4B+ 4GB: https://discuss.moodlebox.net/d/249-apparitiondisparition-du-signal-wifi.

jvonau, 2021-06-02:
>What gets weird is https://wireless.wiki.kernel.org/en/developers/regulatory/crda mentions iw and wpa_supplicant as the means to alter the country code, its been awhile I can't recall if hostapd can alter the country code when starting and wpa_supplicant is not running.

I suspect that hostpd alone can not set the country code properly, check with `iw reg get`. Think that kind of prove my recollection that wpa_supplicant or some hacky script to call iw is needed. As a test on the affected installation I would # out the nohook line in dhcpcd.conf, edit both /etc/wpa_supplicant/wpa_supplicant.conf and /etc/hostapd/hostapd.conf to contain the same country code then reboot. Once the machine is back up check with `iw reg get` and then try the wifi connections. `ps -AH` is useful to see the starting order if the services. Be aware there are other factors that may cause the wifi flakiness such as hdmi resolution or usb3 shielding of cables/connectors to external storage devices. Hope this helps narrow down the cause for you.

martignoni, 2021-11-25:
This issue is not relevant anymore, since MoodleBox 4.0.0 is out and the wireless chip firmwares were upgraded anyway. Closing.

closed 2021-11-25 by martignoni