moodlebox/moodlebox #204: Prevent wpa_supplicant to run

martignoni, 2021-04-20:
As reported by @jvonau: "`wpa_supplicant` is started anyway by `dhcpcd` via `/lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant`". See https://github.com/moodlebox/moodlebox/issues/199#issuecomment-823546160.

We should stop this.

martignoni, 2021-05-12:
@jvonau: so you mean I can let it run without experiencing problems with `hostapd`?

jvonau, 2021-05-12:
I've messed with this networking stuff at [iiab](https://github.com/iiab/iiab) over the years, in the latest iteration of RpiOS out of the box the WiFi chip is rfkill soft blocked unless you run raspi-config or provide a suitable wpa_supplicant.conf with the country code included placed in the root of the first partition, like touch ssh prior to the booting the new image. The user provided wpa_supplicant.conf is moved into /etc/wpa_supplicant/ then the first-boot code notes that the country code is set and unblocks the chip with rfkill, that is how the headless Zero W wifi stuff is intended to work. Without having any upstream connection info included the country code is set but otherwise no control of the WiFi device is taken and hostapd will take control of the device, just have to make sure the country code is the same in both services and things should just work. I have other reasons for keeping wpa_supplicant operational as I have implemented #77

martignoni, 2021-05-14:
Thank you. I think I understand a bit better; l'll leave then the `wpa_supplicant` service running, since it doesn't cause any problems.

martignoni, 2021-05-17:
Works correctly, closing. Thanks @jvonau for the hints!

closed 2021-05-17 by martignoni