moodlebox/moodlebox #213: Test Raspberry Pi OS with Debian Bullseye on its release

martignoni, 2021-07-12:
Debian Bullseye's release is [scheduled for 2021-07-31](https://release.debian.org/). 

Let's test Raspberry Pi OS as soon as it is released with Debian Bullseye

ralf-krause, 2021-08-01:
I'm looking forward to the new Raspberry Pi OS with Debian Bullseye. Will this Raspberry Pi OS automatically work with 64bit? Will it use php with 64bit? Do you have a MoodleBox test image with Debian Bullseye?

martignoni, 2021-08-01:
Me too. All this depends on the official Raspberry OS images.

martignoni, 2021-08-15:
Updates needed to be managed manually:
- php 7.3 -> php 7.4
- mariadb 10.3 -> 10.5

martignoni, 2021-08-15:
TODO: check every modified configuration file for potential change in default values.

E.g. dnsmasq, nginx, mariadb, php, etc.

martignoni, 2021-08-15:
Preliminary version seems to work correctly :-D

martignoni, 2021-08-21:
Build on [branch `WIP-213-master_bullseye`](https://github.com/moodlebox/moodlebox/tree/WIP-213-master_bullseye) works correctly on bullseye (updated from buster).

martignoni, 2021-08-22:
Test image available here: https://moodlebox.net/moodlebox-bullseye-test.img.bz2.

martignoni, 2021-10-16:
Test successful! Closing. Follow up on #210.

closed 2021-10-16 by martignoni