moodlebox/moodlebox #250: User moodlebox is automatically logged in terminal

ralf-krause, 2022-08-04:
There is an automatic login when you have a direct access to the Raspberry Pi.
This problem exists since the user pi is removed with MoodleBox release 4.3.0.
Sorry that I did not post this issue earlier.

### To Reproduce
Connect the Raspberry Pi to a monitor and power on the MoodleBox. 
You will be automatically logged in as the user moodlebox. 
You do not need to write any username or password.

## Additional context
- Raspberry Pi 4B
- MoodleBox 4.3.0
- MoodleBox plugin 2.13.0

martignoni, 2022-08-04:
I'll have to test and confirm this, as I never connect a monitor to my MoodleBoxes 😄

martignoni, 2022-08-04:
Can you please confirm that **you did not** change the auto-login configuration via raspi-config?

I just myself confirmed that the configuration doesn't allow auto-login by default.

ralf-krause, 2022-08-05:
I just copied a new microSD with the MoodleBox Image 4.3.0. I started the MoodleBox and the user moodlebox is automatically logged in. I tested the same with the MoodleBox image 4.2.0 and I get the login prompt.

In the MoodleBox software 4.2.0 I could find the moodlebox user in the passwd but no pi user. Okay, you are renaming the preinstalled pi user. In the release notes for 4.3.0 you write "Remove renaming pi user logic (issue https://github.com/moodlebox/moodlebox/issues/247)." 

Because there is no preinstalled pi user anymore in the current Raspberry Pi OS you need to add the user moodlebox. It is no renaming anymore. It is an adding. And this user should not be logged in automatically.

I hope we get it.
Ralf

martignoni, 2022-08-05:
With a fresh MoodleBox 4.3.0 image, what's the output of the following command:
```
sudo cat /etc/systemd/system/getty@tty1.service.d/autologin.conf
```

PS. Autologin isn't related with #247.

ralf-krause, 2022-08-05:
This is MoodleBox version 4.3.0, 2022-04-18.

MoodleBox is made by Nicolas Martignoni, https://github.com/martignoni/.
Please read the documentation at https://moodlebox.net/.

Last login: Fri Aug  5 09:17:13 2022
moodlebox@moodlebox:~ $ sudo cat /etc/systemd/system/getty@tty1.service.d/autologin.conf
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin moodlebox --noclear %I $TERM
moodlebox@moodlebox:~ $

martignoni, 2022-08-05:
Confirmed. To fix this in your setup, disable autologin with `raspi-config`: System options > Boot / Auto Login > Console (B1).

martignoni, 2022-08-05:
New development image doesn't show this behaviour. So this is fixed.

PS. This bug is somewhat strange, as I didn't enable autologin to the console and the default setup of RPiOS image comes with autologin disabled. Anyway, this will be fixed in the next image.

ralf-krause, 2022-08-05:
Thanks. I fixed the settings im my full installed MoodleBox and it works. Autologin is disabled in raspy-config now.

```
moodlebox@moodlebox:~ $ sudo cat /etc/systemd/system/getty@tty1.service.d/autologin.conf
cat: /etc/systemd/system/getty@tty1.service.d/autologin.conf: No such file or directory
moodlebox@moodlebox:~ $ 

```

martignoni, 2022-08-08:
Related issue: https://github.com/RPi-Distro/pi-gen/issues/614#issuecomment-1117572968.

martignoni, 2022-08-08:
Todo during the build (taken from raspi-config):
```
systemctl --quiet set-default multi-user.target
rm -f /etc/systemd/system/getty@tty1.service.d/autologin.conf
```
and reboot.

closed 2022-08-10 by martignoni