moodlebox/moodlebox #251: Terminal image looks bad

ralf-krause, 2022-08-04:
When you start your MoodleBox with a direct access the Raspberry image looks bad. (see screenshot 1)
I would say that the image is shown in a flat colour version.
Perhaps the image uses colours which are not allowed in terminal.
I can see the four raspberry symbols at the top of the monitor when the Raspberry Pi starts up.

Yes, I can see the correct image when I connect via ssh from my iMac. (see screenshot 2)

For my iPhone and the app Termius the terminal image seems to be too big. (see screenshot 3)

### To Reproduce
Start your MoodleBox with a Monitor connected

### Additional context
- Raspberry Pi 4B
- MoodleBox version 4.3.0
- MoodleBox plugin version 2.13.0
- Monitor iiyama Prolite 1904S

### Screenshots
[image]

[image]

[image]

ralf-krause, 2022-08-05:
I installed the MoodleBox image 4.1.0 on a new microSD because the terminal graphic was changed on Jan 11th, 2022. This was a few days after 4.1.0 was released. So with MoodleBox 4.1.0 I can show the older terminal graphic on a some devices.

Screenshots fro different devices
1 - MoodleBox with monitor iiyama ProLite P1904S
2 - iPad with the Termius app
3 - iPhone X with the Termius app

[image]

[image]

[image]

ralf-krause, 2022-08-05:
Okay, I found that I can change the character size in the Termius app on the iPhone. 
Now I can see the correct terminal graphic without automatic line breaks but the characters are too small on the display.

[image]

martignoni, 2022-08-06:
> I installed the MoodleBox image 4.1.0 on a new microSD because the terminal graphic was changed on Jan 11th, 2022. This was a few days after 4.1.0 was released. So with MoodleBox 4.1.0 I can show the older terminal graphic on a some devices.

Screenshots with logo of current version would be useful.

ralf-krause, 2022-08-06:
I send a lot of screenshots with the current logo. 

There are 4 screenshots in the first message of this thread #251. The first one shows the terminal image when the MoodleBox is connected to my monitor iiyama PreLite P1904S. It could be that the normal terminal console only shows 16 colours. 

The monitor seems to be okay for deeper colours because I can start the same Raspberry Pi 4B with the Raspberry Pi OS full version without any restrictions.

[image]

martignoni, 2022-08-06:
Thanks for the confirmation that the problem occurs in console. I'll have a look to see if it depends on the monitor.

ralf-krause, 2022-08-06:
Could it be that it's a setting in the /boot/config.txt?
The HDMI cable is connected with a DVI-to-HDMI-Adapter.

martignoni, 2022-08-17:
Finally I got my hands on a monitor. Confirming the problem.

closed 2024-02-23 by martignoni