moodlebox/moodlebox #255: Raspberry Pi OS version in the release notes

ralf-krause, 2022-08-15:
It would be useful to write the Raspberry Pi OS version into the release notes of every MoodleBox image.

When an user wants to use a Raspberry Pi Zero W he must know that MoodleBox 4.2.0 is currently the last image which he can use because MoodleBox 4.4.0 is 64bit software.

martignoni, 2022-08-15:
Or maybe should I not support Raspberry Pi Zero W: anyway, performance on it is miserable and I think unusable in production.

What do you think?

ralf-krause, 2022-08-15:
You should not support the Raspberry Pi Zero W. 

Therefore the text https://moodlebox.net/en/help/hardware-needed/ should get a sentence "The Raspberry Pi Zero W could also run the MoodleBox software but performance on it is miserable and I think unusable in production."
And then the Raspberry Pi Zero W should be deleted in all other documents and plugin strings. Yes, the plugin can identify the Raspberry Pi Zero W hardware. But in the strings for the correct power supply we should delete it.

ralf-krause, 2022-08-15:
What must be done to start the MoodleBox 4.4.0 on a Raspberry Pi Zero W?

Must I edit the /boot/config.txt and set arm_64bit=0 and that's all? Then we should take this into the documentation.

ralf-krause, 2022-08-15:
I tested MoodleBox 4.2.0 with a a Raspberry Pi Zero W .... it's very sloooooow.

martignoni, 2022-08-15:
Indeed 👍