moodlebox/moodlebox #259: Install a cloning program

martignoni, 2022-08-29:
See https://discuss.moodlebox.net/d/312-install-a-cloning-program-into-the-moodlebox.

Possible candidates:
- [rpi-clone](https://github.com/billw2/rpi-clone)
- [Image File Utilities](https://forums.raspberrypi.com/viewtopic.php?t=332000)
- Others?

ralf-krause, 2022-10-23:
Has the "Raspberry Pi SD Card Copier" a terminal part which runs without the GUI? Yesterday I just copied a running sd card with 64 GB to a sd card with 32 GB. Yes, it worked fine!

martignoni, 2022-10-23:
> Has the "Raspberry Pi SD Card Copier" a terminal part which runs without the GUI?

Raspberry Pi Imager (if this is what you mean) has a CLI, but its installation forces a lot of useless GUI things, weighting about 313 MB. That's to much.

ralf-krause, 2022-10-24:
No, I do not mean the Raspberry Pi Imager. 

There is also a program which has the name "SD card copier". This program can copy the running sd card to an usb card reader, to an usb stick or an usb ssd. It is also possible to copy a running sd card with 64 GB on an usb card reader wit a smaller sd card with 16 GB.

martignoni, 2022-10-24:
> There is also a program which has the name "SD card copier".

Ok, found the source is here: https://github.com/raspberrypi-ui/piclone, but it's even worst: around 552 MB (and it is GUI only).

martignoni, 2022-10-25:
[rpi-clone](https://github.com/billw2/rpi-clone) works correctly and is very light. It will be integrated into the next MoodleBox image.

martignoni, 2022-10-25:
This works ok. Closing.

closed 2022-10-25 by martignoni