moodlebox/moodlebox #271: Android and the mobile Moodle app

ralf-krause, 2022-10-23:
Do you have in mind that Android does not like to use self signed certificates?
https://discuss.moodlebox.net/d/254-mobile-app/19

We discussed the problem in the international Moodle forum "MoodleBox and mobile Moodle app". You also saw the answer from Juan Leyva about the problem. The mobile app will only work with an Android device when the ssl part is switched of in the nginx configuration `/etc/nginx/sites-available/default`.
https://moodle.org/mod/forum/discuss.php?d=424353#p1747266

martignoni, 2022-10-23:
Thanks for the heads up, I had completely forgotten about it.

martignoni, 2022-10-23:
I'll fix this by
- commenting out SSL enabling lines in [default Nginx configuration](https://github.com/moodlebox/moodlebox/blob/main/roles/webserver/templates/etc/nginx/sites-available/default.j2), and
- adding a step in the [how-to switch MoodleBox to HTTPS](https://moodlebox.net/en/help/connexion-https/) documentation.

Anyway, people who want HTTPS have to enter CLI mode, so another step is not so blocking.

Ref: https://github.com/moodlebox/moodlebox.net/issues/59

martignoni, 2022-10-25:
Fix is included. Closing.

closed 2022-10-25 by martignoni