moodlebox/moodlebox #273: Error when updating to Moodle 4.0.4+ (20221108)

ralf-krause, 2022-11-08:
I got an error when I updated up to Moodle 4.0.4+ (20221108).

`Coding error detected, it must be fixed by a programmer: Cannot set custom filetypes because they are defined in config.php`

I think that the error is specific for the MoodleBox. I updated two other platforms to the same Moodle version. On my MoodleBox it is not possible to open the web administration.

After I set the parameter customfiletypes to xcustomfiletypes I can open the web administration again. Yes, I know that it does not make sense to change a parameter name. I did this to find a solution for this problem.
```
$CFG->xcustomfiletypes = array(
  (object)array(
    'extension' => 'crt',
    'icon' => 'sourcecode',
    'type' => 'application/x-x509-ca-cert',
    'customdescription' => 'X.509 CA certificate'
  )

```

Ralf

martignoni, 2022-11-08:
This is very annoying. I've to investigate why the update process has to set custom file types.

ralf-krause, 2022-11-09:
Thanks for your answer!
I created an issue in the Moodle tracker https://tracker.moodle.org/browse/MDL-76247

abias, 2022-11-09:
Hi all,

in the end, this issue turned out to be a bug in theme_boost_union which @ralf-krause uses in his moodlebox.
The bug and the fix is explained on https://github.com/moodle-an-hochschulen/moodle-theme_boost_union/issues/133.

This issue here can be closed then.

closed 2022-11-09 by martignoni

ralf-krause, 2022-11-09:
I just upgraded the plugin Boost Union on my MoodleBox. The MoodleBox runs fine again.