moodlebox/moodlebox #283: Remove useless build options

martignoni, 2022-12-29:
A few build options do not make sense anymore:
- `moodlebox_set_hostname`
- `moodlebox_set_timezone`
- `moodlebox_set_locale`
since we want them always enabled.

martignoni, 2022-12-30:
Works as expected, closing.

closed 2022-12-30 by martignoni

ralf-krause, 2023-02-14:
Did you remove these parameters for the image moodlebox-4.6.0-rpi-imager.img.xz only? Or for both images? 

I think the parameters are needed for the image moodlebox-4.6.0.img.xz because this image will not be configured by the extended settings of the Raspberry Pi Imager. This image should be copied by every other copy tool like Etcher.

martignoni, 2023-02-14:
Only the capacity to set different values when **building** the image were removed (see code). The parameters are of course present, otherwise MoodleBox could not start.