moodlebox/moodlebox #285: Enable smartphone tethering out of the box

martignoni, 2023-02-11:
Some project already does it, e.g.
- https://github.com/openplotter/openplotter-network/tree/master/openplotterNetwork

ralf-krause, 2024-04-01:
The smartphone tethering via wifi works fine with the AP+STA mode since the MoodleBox-4.0.0. 

I tested the smartphone connection for my post https://discuss.moodlebox.net/d/451-connecting-the-moodlebox-to-a-wifi-network . We do not need any additional hardware ... no special wifi dongle, no special usb cable and no exclusive adapter. And we do not need any additional software. It works with the Raspberry Pi wifi driver. 

The current problem is that the smartphone tethering via wifi does not work with MoodleBox-4.6.0 and 4.7.0 (both are running with Bookworm). The automatic wifi connection works and also getting the internet time. But we do not get intenet pages or Moodle updates. I think it should be easy to fix.

martignoni, 2024-04-01:
> The smartphone tethering via wifi works fine with the AP+STA mode since the MoodleBox-4.0.0.

Just to be clear: this is not the purpose of this improvement: AP+STA (which enables WiFi tethering) works out of the box since 4.0.0.

I just updated the description for clarity: the scope of this enhancement is tethering **via USB**.

martignoni, 2025-06-07:
Testing with iPhone:
```
moodlebox@moodlebox:~ $ ip a
...
6: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d2:88:0c:36:34:05 brd ff:ff:ff:ff:ff:ff
    inet 172.20.10.5/28 brd 172.20.10.15 scope global dynamic noprefixroute eth1
       valid_lft 3441sec preferred_lft 3441sec
    inet6 fe80::2941:91af:f4e6:7fa6/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

moodlebox@moodlebox:~ $ ip route show
default via 192.168.1.1 dev eth0 proto dhcp src 192.168.1.150 metric 100 
default via 172.20.10.1 dev eth1 proto dhcp src 172.20.10.5 metric 101 
default via 10.0.0.1 dev uap0 proto static metric 600 
10.0.0.0/24 dev uap0 proto kernel scope link src 10.0.0.1 metric 600 
172.20.10.0/28 dev eth1 proto kernel scope link src 172.20.10.5 metric 101 
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.150 metric 100 
```

martignoni, 2025-06-07:
Testing with Android device:
````
moodlebox@moodlebox:~ $ ip a
...
7: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether a2:df:c3:07:d4:ad brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.241/24 brd 192.168.42.255 scope global dynamic noprefixroute usb0
       valid_lft 3527sec preferred_lft 3527sec
    inet6 fe80::8f61:71e:c325:ba2/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

moodlebox@moodlebox:~ $ ip route show
default via 192.168.1.1 dev eth0 proto dhcp src 192.168.1.150 metric 100 
default via 192.168.42.129 dev usb0 proto dhcp src 192.168.42.241 metric 101 
default via 10.0.0.1 dev uap0 proto static metric 600 
10.0.0.0/24 dev uap0 proto kernel scope link src 10.0.0.1 metric 600 
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.150 metric 100 
192.168.42.0/24 dev usb0 proto kernel scope link src 192.168.42.241 metric 101 
```

martignoni, 2025-06-07:
Tested successfully with an iPhone. Impossible to test with an Android smartphone: I don't own such a device. Anyway, the tethering seems to work OK, so marking as tested.

closed 2025-06-07 by martignoni