moodlebox/moodlebox #286: PHP 8 for the MoodleBox

ralf-krause, 2023-02-19:
When do you plan to update the MoodleBox to PHP 8.0? 
The security support for PHP 7.4 ended in November 2022.

martignoni, 2023-02-19:
PHP8.2 will be part of Debian Bookworm due later this year. We can expect Bookworm will be the base of a future RpiOS image, and MoodleBox will then use it (just like for #129 for Buster in 2019 and #210 for Bullseye in 2021).

Meanwhile, there is no need to worry too much about safety of PHP7.4 in Debian, because the maintainers of the package will contenue security support on their own. See [this post for why](https://unix.stackexchange.com/a/718282).

martignoni, 2023-02-20:
Closing in favor of #287 (update to Debian Bookworm).

closed 2023-02-20 by martignoni

martignoni, 2023-02-20:
Another important thing (from https://moodledev.io/general/development/policies/php#php-versions-under-development):
> PHP 8.2 support is currently being implemented. We hope to make this available for Moodle 4.2 but this has not yet been confirmed and is subject to change. See [MDL-76405](https://tracker.moodle.org/browse/MDL-76405) for details.