moodlebox/moodlebox #321: Update GPU memory setting to current spec

martignoni, 2024-01-03:
GPU memory setting has changed as of latest Raspberry OS version. See the [official doc](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#gpu_mem).

ralf-krause, 2024-02-24:
Does the MoodleBox need any GPU memory or should the setting get a minimum memory. The GPU memory seems to be part of the normal RAM so minimizing GPU should leave more RAM for other processes. Correct?

martignoni, 2024-02-24:
The GPU mem should be minimal (we use almost no any graphic power).

But in recent version of RPiOS, this setting is not used anymore, and I'll drop it probably.

ralf-krause, 2024-02-24:
Does the current RaspberryPiOS genarally use no GPU memory size or only for Raspberry Pi 5?

martignoni, 2024-02-24:
See the [official doc](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#gpu_mem), which is not very specific about this. That's why I'll probably won't change anything: if it ain't broken, don't fix it.

ralf-krause, 2024-02-24:
As I understand the official doc the Raspberry Pi does not allocate  any GPU memory by the os. For all other Raspberry Pi models the config parameter gpu_mem is used for setup. "To ensure the best performance of Linux, you should set gpu_mem to the lowest possible value." Could the lowest value be zero?

martignoni, 2024-02-24:
Anyway, second to the [official doc](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#gpu_mem), gpu_mem setting is obsolete and seems to be useful only for legacy kernels.

martignoni, 2024-03-07:
As comment https://github.com/moodlebox/moodlebox/issues/321#issuecomment-1962520904, not needed.

closed 2024-03-07 by martignoni