moodlebox/moodlebox #323: Database collation should be utf8mb4_unicode_ci

ralf-krause, 2024-02-22:
## Describe the bug
The MoodleBox database collation is utf8mb4_general_ci and should be utf8mb4_unicode_ci instead.
Please look at https://docs.moodle.org/403/en/MySQL_full_unicode_support

### To Reproduce
You can see the database collations when you get http://moodlebox.home/adminer.php
And you should see the database collation in the Moodle config.php

## Expected behavior
I think your setting with the collation utf8mb4_general_ci should be no problem as long as nobody gets a collation mix in the database. I heart about the mix problem in https://moodle.org/mod/forum/discuss.php?d=455362 . I can’t say if the installation of a plugin will be able to add database tables with a different collation.

## Additional context
- Raspberry Pi 4B
- MoodleBox 4.6.0
- MoodleBox plugin 2.17.4
- Moodle 4.3.3+ (20240215)

## Screenshots
No screenshot

martignoni, 2024-02-22:
Did you experience any problems with current `utf8mb4_general_ci` collation? (other than MoodleHQ recommendation)

ralf-krause, 2024-02-22:
As I wrote it should be no problem as long as nobody gets a collation mix in the database. I can‘t say from which reason the database get‘s a collation mix … could it be a plugin installation? Could it be a Moodle course restored from a platform with a different collation?

closed 2024-02-23 by martignoni