moodlebox/moodlebox #334: Can't connect the MoodleBox to the internet via a wifi network

ralf-krause, 2024-04-03:
## Prerequisites
- [ X] I checked the documentation and found no answer
- [ X] I checked to make sure that this issue has not already been filed

## Describe the bug
The MoodleBox does not connect to a wifi network. 
Please look at https://discuss.moodlebox.net/d/451-connecting-the-moodlebox-to-a-wifi-network

I edit the file  `/etc/NetworkManager/system-connections/WifiSTA.nmconnection`, indicating the SSID and password of your wireless access point, then reboot. The MoodleBox connects to the wifi router correctly. The MoodleBox gets an ip address and it can be seen in the router list of connected devices. The MoodleBox startpage and the learning platform are working fine. The system clock is synchronized with the internet time. 

The problem is that I can't get any external url. I try to connect to my router with 2.4G and 5G ... no connection to the internet. And I try to use the smartphone hostspot of my iPhone ... no connection to the internet. I can ping the router and the iPhone but no external address. The wifi connection should work in the same way as the ethernet connection works. 

### To Reproduce
1. Copy a fresh microSD card with MoodleBox-4.7.0.img.xz 
2. Start up the MoodleBox
3. Connect to the MoodleBox wifi
4. Open the browser with http://moodlebox.home ... it should work
5. Open the terminal and connect with moodlebox@moodlebox.home
6. Edit the file `/etc/NetworkManager/system-connections/WifiSTA.nmconnection` and put in your router ssid and password.
7. Reboot the MoodleBox
8. The MoodleBox will connect to your router automatically
9. Reconnect to the MoodleBox wifi
10. Open the browser with http://moodlebox.home ... it should work as before
11. Enter an external url, e.g. https://moodlebox.net
12. The browser cant find the page

## Additional context
- Raspberry Pi 4B (4GB RAM)
- MoodleBox 4.7.0 (same problem with 4.6.0)
- MoodleBox plugin version 2.17.5
- No other changes expect setting the language for the moodlebox user

## Attached files
I made files to get the different configurations: (1) with ethernet, (2) with wifi hotspot, (3) with no internet connection.
The command `nmcli device show > nmcli-device-show-xxxxxx.txt` helps to get the configuration into a text file.

[nmcli-device-show-ethernet.txt](https://github.com/moodlebox/moodlebox/files/14852316/nmcli-device-show-ethernet.txt)
[nmcli-device-show-hotspot.txt](https://github.com/moodlebox/moodlebox/files/14852317/nmcli-device-show-hotspot.txt)
[nmcli-device-show-no-internet.txt](https://github.com/moodlebox/moodlebox/files/14852318/nmcli-device-show-no-internet.txt)

ralf-krause, 2024-04-03:
I test the same setting with MoodleBox-4.5.1 and it works fine.
https://discuss.moodlebox.net/d/451-connecting-the-moodlebox-to-a-wifi-network/7

The Raspberry Pi OS of MoodleBox-4.5.1 is Debian 11 (Bullseye) and the Raspberry Pi OS of MoodleBox-4.7.0 is Debian 12 (Bookworm).

martignoni, 2024-04-05:
This modification to the WifiSTA connection fixes the problem:
```
sudo nmcli con mod WifiSTA ipv4.route-metric 99
```
Reboot after the modification.

ralf-krause, 2024-04-06:
Great! 

Yes, this works for me. The MoodleBox connects to my wifi router and I get webpages from the internet

ralf-krause, 2024-04-06:
Sorry. I was wrong.

I used the microSD card with the MoodleBox-4.6.0. In this case the command `sudo nmcli con mod WifiSTA ipv4.route-metric 99` works fine as we want to get it. 

Now I copy a new microSD card with the MoodleBox-4.7.0. I edit the file `/etc/NetworkManager/system-connections/WifiSTA.nmconnection` and put in your router ssid and password. And then I run the command  `sudo nmcli con mod WifiSTA ipv4.route-metric 99` and it does not work. Sorry for this confusion!

ralf-krause, 2024-04-06:
Stupid thing! Now it works also with MoodleBox-4.7.0. 

The MoodleBox is connected via wifi to my smartphone hotspot. Now I get the internet webpages which I try to get

closed 2024-04-07 by martignoni

ralf-krause, 2024-11-04:
Works fine in MoodleBox 4.8.0. 
Please look at https://discuss.moodlebox.net/d/451-connecting-the-moodlebox-to-a-wifi-network/14

PR #335 (2024-04-07): Change the metric of wireless connection
Fixes #334.