moodlebox/moodlebox #340: Can't set wifi channels 12 and 13 for MoodleBox wifi

ralf-krause, 2024-04-25:
## Prerequisites
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Describe the bug
I want to set the channel for the MoodleBox wifi to 13. The regulatory country is Germany and all channels from 1 to 13 should be allowed. I can set all channels from 1 to 11 but not 12 and 13.

### To Reproduce
1. Go to MoodleBox dashboard
2. Choose the wifi regulatory country to "Germany"
3. Choose wifi channel 13
4. Click on "Change wifi settings"
5. Wait a moment
6. Reload the MoodleBox dashboard
7. The country changed to Germany but the channel is not 13  :-(
8. Take the Android wifi analyser from F-Droid ... the MoodleBox wifi is not on channel 13

## Additional context
- Raspberry Pi model 4B
- MoodleBox version 4.7.0
- MoodleBox plugin version 2.17.5 (the preinstalled plugin version)

## Screenshots
No, I take no screenshot from MoodleBox 4.7.0.
But I tested MoodleBox 4.5.0 ... and in 4.5.0 the channel 13 can be set for MoodleBox wifi.

[image]

ralf-krause, 2024-08-08:
Searching in the source code I found the reason for this issue.

```
def do_channel():
    """Channel setting."""
    global new_channel
    # Validate new_channel. Replace it with default_channel if invalid.
    if int(new_channel) < 1 or int(new_channel) > 13:
        new_channel = default_channel
    # Channel 12 and 13 aren't valid in Canada and US.
    if new_country in ['CA','US'] and int(new_channel) > 11:
        new_channel = default_channel
    # new_channel is now valid.
    if is_networkmanager():
        # Workaround bug in NetworkManager: country_code is not used, so if new_channel is 12 and 13,
        # we revert to default_channel (11).
        # See https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/960.
        new_channel = str(min(int(default_channel), int(new_channel)))
        # Set channel with nmcli
        subprocess.run(['sudo', 'nmcli', 'con', 'mod', 'WifiAP', 'wifi.channel', new_channel])
    else:
        # Set channel in hostapd config file.
        file_replace_line(hostapd_conf_file, 'channel=.*', 'channel=' + new_channel)

```

ralf-krause, 2024-10-27:
I am using MoodleBox 4.7.0 with Moodle 4.4.4+ (20241025).

Today I update the Raspberry OS to the latest version. 
`Linux moodlebox 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64`

Then I edit the file /etc/NetworkManager/system-connections/WifiAP.nmconnection and set the channel to 13. The regulatory country is Germany (DE) ... I can't see the regulatory country in the config file. I want to see if the channel 13 works after a lot of updates. Yes, after a reboot the MoodleBox now starts with channel 13. 

So I hope you can change the restriction in the [changewifisettings.py](https://github.com/moodlebox/moodle-tool_moodlebox/blob/main/bin/changewifisettings.py) now.

```
[connection]
id=WifiAP
uuid=e58aa9e5-979d-4dd9-a2bf-2439fbbb89a3
type=wifi
interface-name=uap0
timestamp=1729942770

[wifi]
band=bg
channel=13
mode=ap
ssid=MoodleBox

[wifi-security]
group=ccmp;
key-mgmt=wpa-psk
pairwise=ccmp;
proto=rsn;
psk=moodlebox

[ipv4]
address1=10.10.10.1/24,10.10.10.1
method=shared

[ipv6]
addr-gen-mode=default
method=shared

[proxy]

```

There are some pictures from my tests.

MoodleBox Dashboard
[image]

macOS Wifi menu (option key pressed)
[image]

Android Wifi Analyser
[image]

martignoni, 2024-10-27:
The [workaround](https://github.com/moodlebox/moodle-tool_moodlebox/blob/36d09dad841b94e37ecbaf99307c0b0103fbd735/bin/changewifisettings.py#L137-L140) in `changewifisettings.py` can be indeed removed, since new versions of RPi OS have solved the problem.