moodlebox/moodlebox #349: AdminerEvo is a newer version of Adminer

ralf-krause, 2024-10-27:
What do you think about AdminerEvo for the MoodleBox?
https://github.com/adminerevo/adminerevo

Adminer is a very nice software but it's no maintain since 3 years. AdminerEvo is using the sources of Adminer. The current version of the Moodle plugin Adminer https://moodle.org/plugins/local_adminer also uses AdminerEvo.

martignoni, 2024-10-27:
Thanks, I didn't know about his maintained fork of Adminer. Investigating now.

martignoni, 2024-10-27:
Download URL: https://download.adminerevo.org/latest/adminer/adminer.zip

martignoni, 2024-10-27:
Direct download from Github (for 4.8.4 version): https://github.com/adminerevo/adminerevo/releases/download/v4.8.4/adminer-4.8.4.php

closed 2024-10-27 by martignoni

ralf-krause, 2024-10-27:
Thanks! This Adminer version works fin with PHP 8.1, 8.2 and 8.3.