moodlebox/moodlebox #354: Fix AP issue with Raspberry Pi Model B

martignoni, 2024-11-01:
Fix for https://github.com/moodlebox/moodle-tool_moodlebox/issues/145 doesn't completely work, as it needs to access to the GUI, which is impossible without wireless connection.

closed 2024-11-01 by martignoni

ralf-krause, 2024-11-02:
The name of this issue should be "Fix AP issue with Raspberry Pi 3B".

ralf-krause, 2024-11-02:
With your fix you will set the wifi security WPA on a Raspberry Pi 3B everytime the system is rebooted.

But if you now move a configured microSD card from a Raspberry Pi 3B into a Raspberry Pi 3B+, 4B, or 5 then the wifi security WPA will not change to WPA2 automatically. The wifi security will only change to WPA2 if someone sets any wifi parameter and the script changewifiparameters.py is executed.

martignoni, 2024-11-02:
> The wifi security will only change to WPA2 if someone sets any wifi parameter

Yes. I think this is acceptable since there's no problem making such change (and this use case is presumably rare).

It is only a workaround, and I hope the problem is fixed upstream, so the whole thing become history.